ronoaldo/FlashforgeFinderIntegration #2: Flashforge Thumbnail to Prusaslicer idea.

KeltE, 2021-02-01:
Hello!
  i am very ignorant of codes. Since I am using Prusaslicer on Flashforge Dreamer NX, the idea came to mind. Post-processing scripts can be added in Prusaslicer. Is there any way to customize your Cura plugin so that the script takes the same thumbnail and simply adds Gcode to the beginning when saving? And the .GX end can be added to the "Output fileneame format" line. 
Best regards Egon

ronoaldo, 2021-02-01:
Hello @KeltE - yes, it could!

I'm not familiar yet with PrusaSlicer coding, but I attempted to make the code here modular, in order to allow these use cases.

I can tune the code that generates **.gx** file to take the image as input. Does Prusaslicer provides some way to export the current view as an image file and pass this data to the post-processing script?

We can try to coordinate with PrusaSlicer maintainers in order to make this use case possible. I find it very, very useful to be able to preview the **gcode** somehow and the embedded image makes a big difference i usage.

KeltE, 2021-02-01:
As I understand it, Prusaslicer generates a thumbnail inside the G code. All you have to do is set the desired size in the Printer settings> General> G-code Tumbnail box. https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/way-to-enable-prusaslicer-thumbnails-for-mk3s/
 In the beginning, if I didn't know that Marlin flavor was running on Dreamer Nx, I used the Perl script found on Uptimfab.com to replace the wrong warm-up commands.
In Prusaslicer you can set the G code output format you want. In principle, it is not even necessary to make the .GX format separately. If you could only get a thumbnail with Script at the beginning of the program. 
[image]

ronoaldo, 2021-02-02:
I can take a look at how that works out, but if the image is already embedded then the python script will be able to convert from base 64 to the .gx binary header.

KeltE, 2021-02-02:
Thank you very much!

ronoaldo, 2021-02-02:
For the record, there are issues with Prusa printers, and it makes more sense on why Flashforge picked a binary, one-line, grayscale BMP. Anyway, based on the comments from @KeltE and the discussion here, it should be doable: https://github.com/prusa3d/PrusaSlicer/issues/4127

KeltE, 2021-02-03:
I will add a Gcode to a cube made with one Prusaslicer. You mentioned in the plugin data that the thumbnail size is 80x60. At the beginning of the G code is a thumbnail generated by Prusaslicer. Maybe it will help.
[xyzCalibration_cube_0.2mm_PLA_44m.txt](https://github.com/ronoaldo/FlashforgeFinderIntegration/files/5917178/xyzCalibration_cube_0.2mm_PLA_44m.txt)