ronoaldo/FlashforgeFinderIntegration #3: GXplugin script for a two-nozzle printer.

KeltE, 2021-04-08:
Hello!
 Because on dual-nozzle printers, Cura adds the Gcode T0 command before it starts, and the printer recognizes it as an invalid code. Is it possible to make a script that checks before the start of Gcode T0 for existence and if necessary removes it? And would that be possible at all?
 Many thanks! Egon

ronoaldo, 2021-04-20:
Hi KeltE Did you already contributed a fix for this or you need help with the gcode manipulation?

KeltE, 2021-04-20:
Hello! Unfortunately, I do not have the appropriate skills. Then yes please help. I made a Flashforge dreamer NX for the Cura profile which is now part of Cura 4.9. And as a side project, I also made a profile concept for Flashforge Dreamer. As testing revealed, Cura adds the T0 command to the G-code of the two-tool printer before starting G-code.
The GXwriter plugin in your profile also works with other Flashforge printers, so the idea came that maybe it is possible to add a snippet of code that checks and, if necessary, removes the T0 command from the beginning of the program.

**Flashforge Dreamer Gcode start With my profile.**
;TIME:5212
;Filament used: 2.23152m, 0m
;Layer height: 0.1
;MINX:-14.113
;MINY:-22.05
;MINZ:0.3
;MAXX:22.023
;MAXY:22.05
;MAXZ:34.6
;Generated with Cura_SteamEngine 4.8.0
**T0**                                                                                              **(This is te problem line)**
M82 ;absolute extrusion mode
;Start Gcode
G90 ;absolute positioning
M140 S60 T0 ;Heat bed up to first layer temperature
M104 S200 T0 ;Set nozzle temperature to first layer temperature
M107 ;start with the fan off
G90
G28
M132 X Y Z A B ;Load Offsets From EEPROM
G1 Z50.000 F420
G161 X Y F3300 ;Move Axes to Maximum
M7 T0
M6 T0
M651
M907 X100 Y100 Z40 A80 B80 ;Digital potentiometer value
M108 T0 ;Select Active Extruder
;Purge line
G1 X-110.00 Y-60.00 F4800
G1 Z0.3 F420
G1 X-110.00 Y60.00 E17,4 F1200
;Purge line end

Many thanks!

ronoaldo, 2021-04-20:
The printer you have supports/uses the .gx file extension? If yes, I can try to detect this instruction when you select the .gx mode and drop that particular g-code.

ronoaldo, 2021-04-21:
Sorry @KeltE I see you already can work with the GXWriter and the other printers. That is great then! I can detect for this model and make it work. For me to test can you tell me how I add the printer in Cura? If you have the profiles I apreciate!

KeltE, 2021-04-21:
Since Dreamer NX is part of Cura 4.9 then I have other profiles made around it. Must copy the Dreamer profile on my github page to the Recourses directory. And your profile may no longer be visible under Cura 4.9 at Flashforge. I recommend changing the manufacturer's name a bit (for example Flashforge1).

ronoaldo, 2023-04-11:
Hi @StevenVDurm , unfortunately no. I had done some planning but could not start coding/testing as my Finder got a nozzle issue, so I'm avoiding doing changes before fixing that part.

StevenVDurm, 2023-04-11:
Thank you for letting me know.
If you need to do some testing for the flashforge Dreamer, then I'll gladly try it out for you.