ronoaldo/FlashforgeFinderIntegration #4: Cura 4.9.0

janagn, 2021-04-20:
Hello,

I was wondering if there are any specific instruction for Cura 4.9.0. Downloaded from Marketplace seems to do nothing and dropping the packages to Cura also seems to do nothing.

Kind Regards
Ioannis

ronoaldo, 2021-04-20:
Hi @janagn - I tested on Cura 4.9 today. The way this works is:
- Version 1.0.0 (current from Marketplace): you need to first install the printer files from Plugins -> Flashforge
- Version 1.0.1 (under review from Marketplace): you don't need to do any extra steps

After either 1 or 2 is done, you need do add a new printer with the new printer dialog, and look for Flashforge -> Flashforge Finder.

Can you check please if this enables you to add the Finder printer and use the shipped profiles?

ronoaldo, 2021-04-20:
PS: if you pick from Releases and download the .curapackages for 1.0.1 you should be able to easily add the printer after restart!

janagn, 2021-04-20:
Hi and thank you for the quick relay. I manage only to install manually the 1.0.1 GX writer by downloading the .curapackage, I opened it with 7-zip and exported its contents in the plugin folder, That worked as expected.

For flash forge finder I followed the same procedure but I still cannot see the printer in the list of FlashForger PRinters.

(Version 1.00 I gave up as I downloaded it from the market place and didn't show at all)

ronoaldo, 2021-04-21:
@janagn thanks for the print page. I just tested with a clean install of the 1.0.1, by drag'n'drop both .curapackages and after restart, cura shows the Finder under Flashforge vendor when adding a printer:

[image]

To use the version from Marketplace (1.0.0) you need more steps (because of the way I implemented initially):

- Open Cura
- Install from Marketplace
- Restart
- Go to Extensions -> Flashforge Finder -> Install support files
- Restart again
- Add new printer has the Flashforge Finder as an option under the Flashforge maker (like screenshot above)

janagn, 2021-04-21:
Hi Ronaldo,

Thank you for the details explanation. So, ver 1.0.1 when I drag and dropped on the Cura 4.9.0 workspace did work as you explained. For ver 1.0.0 though, installing from marketplace, I never get the option in the Extensions menu both in Win64 and Linux. 

Kind Regards
Yiannis

ronoaldo, 2021-04-21:
Thanks @janagn - soon the 1.0.1 version will land in the Marketplace with the fix. I'll keep this open until that. Thanks for the report!

KeltE, 2021-04-21:
> @janagn thanks for the print page. I just tested with a clean install of the 1.0.1, by drag'n'drop both .curapackages and after restart, cura shows the Finder under Flashforge vendor when adding a [...]

It's good to see that your profile also shows. When I tested the Dreamer NX profile on Cura 4.8, for some reason I couldn't show other profiles other than by changing the manufacturer.

commit 9e911b70 (2021-04-28): Renamed definitions to fix marketplace release.

Currently, publishing a definitions folder as part of
your addon may result in a bug that is being worked
on by Ultimaker team.

This patch potentially closes #4 as it will no trigger
the add-on publishing bug, and allows the release to
be properly worked on.
closed 2021-04-28 by ronoaldo