ronoaldo/FlashforgeFinderIntegration #6: Quality profiles not available in Cura 4.5 and earlier

Ghostkeeper, 2021-04-28:
Hi, I've just reviewed and tested your plug-in as you submitted it to the Marketplace. And I found a bug.

The printer profiles include some quality profiles (low, standard, high and hyper). However these profiles seem to be made for Cura 4.6. Earlier versions of Cura can't downgrade them. As a result, the profiles aren't available and it loads the default set of profiles or no profiles at all.

The following errors are put in the log:

```
2021-04-28 11:56:25,454 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container finder_standard: {'quality_type': 'normal', 'type': 'quality', 'name': 'Standard Resolution', 'weight': '0', 'version': '4', 'definition': 'finder', 'container_type': <class 'UM.Settings.InstanceContainer.InstanceContainer'>, 'global_quality': 'True', 'setting_version': '13', 'id': 'finder_standard'}
2021-04-28 11:56:26,084 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container finder_hyper: {'quality_type': 'hyper', 'type': 'quality', 'name': 'Hyper Resolution (Slower)', 'weight': '0', 'version': '4', 'definition': 'finder', 'container_type': <class 'UM.Settings.InstanceContainer.InstanceContainer'>, 'global_quality': 'True', 'setting_version': '13', 'id': 'finder_hyper'}
2021-04-28 11:56:26,308 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container finder_low: {'quality_type': 'low', 'type': 'quality', 'name': 'Low Resolution (Faster)', 'weight': '0', 'version': '4', 'definition': 'finder', 'container_type': <class 'UM.Settings.InstanceContainer.InstanceContainer'>, 'global_quality': 'True', 'setting_version': '13', 'id': 'finder_low'}
2021-04-28 11:56:26,703 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [324]: Invalid metadata for container finder_high: {'quality_type': 'high', 'type': 'quality', 'name': 'High Resolution (Slow)', 'weight': '0', 'version': '4', 'definition': 'finder', 'container_type': <class 'UM.Settings.InstanceContainer.InstanceContainer'>, 'global_quality': 'True', 'setting_version': '13', 'id': 'finder_high'}
```

They are invalid because it expects (in this case) the `setting_version` metadata to be 11.

Setting the `setting_version` to 6 should allow the profiles to be loaded in Cura 4.0 and higher, which is what the plug-in is marked to support. You should then be careful not to change any settings that are implemented in those newer Cura versions. I think the best approach would be to just set them to 6 and test it on all supported Cura versions. In Cura 4.0, see if it produces any errors in the log about settings that don't exist, and if so remove those setting overrides.

If you really need to use features of more recent Cura versions, you'd have to modify the installer to install different profiles in modern Cura versions than in the old ones. That would get complex.

ronoaldo, 2021-04-29:
If I was to update the plugin to be compatible with setting_version 11 or newer, what changes I need to do in the plugin/Marketplace submission?

I ask this because I may not have time to test the profiles and make test prints on all previous Cura versions.

Ghostkeeper, 2021-05-13:
> If I was to update the plugin to be compatible with setting_version 11 or newer, what changes I need to do in the plugin/Marketplace submission?
>
> I ask this because I may not have time to test the profiles and make test prints on all previous Cura versions.

Understandable!

You'll want to submit your plug-in as using SDK version 7.2.0 then. That SDK version was introduced with Cura 4.6, so using that SDK version indicates that you support Cura 4.6 and later, but not 4.5 or earlier. It will not show up in the Marketplace for older Cura versions.

ronoaldo, 2021-05-19:
@Ghostkeeper thanks for the heads up. If you have some spare time, can you review if I did it right at this PR #8?

If it's all good, I'll try to re-upload the version on Marketplace so it prevents unexpected behavior from users in Cura <= 4.5.

Ghostkeeper, 2021-06-02:
I've reviewed them. I suggested one change. Sorry that it took a while.

ronoaldo, 2021-06-26:
Merged, releasing minor update soon.

closed 2021-06-26 by ronoaldo