ronoaldo/FlashforgeFinderIntegration #10: [Question] Using USB to monitor and send codes

ElSaam, 2021-07-05:
Is is possible to utilize the usb port and drivers to monitor and print from Cura?

I'm no professional or anything and I'm asking so that other people wondering can see the answer here.

ronoaldo, 2021-07-19:
Hi @ElSaam currently, I don't have support for this but it is planed for the future. I was about to start with wi-fi, which is what I use, but USB basically uses the same communication over a different connection/socket.

ElSaam, 2021-07-24:
> Hi @ElSaam currently, I don't have support for this but it is planed for the future. I was about to start with wi-fi, which is what I use, but USB basically uses the same communication over a different connection/socket.

Wow that's amazing! Such a good motivation for an uncommon printer.  So glad we have you working on it 😄.

closed 2021-07-24 by ElSaam