ronoaldo/FlashforgeFinderIntegration #11: Plugin / CURA / Octoprint Integration

OGYardens, 2021-10-04:
Good Day;

I was able to get this installed and change the settings to match the Flashforge Creator Max.  I have not yet tried using it to print, but ran into an issue where after adding this plugin I am no longer able to connect to the Raspberry Pi with Octoprint on it for online management.

Would you know what I can do to help verify this plugin is not interfering with the Octoprint connection?  It was working until this plugin was activated.  Could be coincidental, but I want to be sure.

ronoaldo, 2021-12-16:
Hi @PHiGGYsMALLS  - sorry for the very late reply. It can be something with my plugin, specially if it was working before you installed. I can try to help you troubleshoot but I would need to know what Octoprint connection you are having issues to try to replicate them:

- What is the Octoprint plugin you are using?
- What version of Cura you have?

Also, if you remove this plugin, will the issue go away?

PHiGGYsMALLS, 2021-12-17:
Hi @ronoaldo , please see below replies

> * What is the Octoprint plugin you are using?
I acquired a new computer and don't know what plug in it was any more. 

> * What version of Cura you have?
I am currently using version 4.12.1 and have not had success in getting any plugins to work with the Creator Max / Octoprint with any success; no ability to connect and send prints.  
> 
> Also, if you remove this plugin, will the issue go away?
I have since abandoned any attempts to use an Octoprint plugin for the Creator Max because it is too high maintenance and problematic.  I went back to using Flashprint.

Thank you,

Jan

closed 2023-03-21 by ronoaldo