ronoaldo/FlashforgeFinderIntegration #12: Platform back to home position

yvaha, 2021-12-25:
Great work: Extension makes it possible to print using Cura on Raspberry Pi 4.

Only remark: after printing the printhead is brought to the home position and platform does not go back to home position.

For protection of the print nozzle, the platform should first go back to the home position. After that print head can go back to home position.

ronoaldo, 2021-12-27:
Hi @yvaha thanks for the feedback!

Just so I understand the issue with the homming after print completes: you sliced the print with the printer settings into Octoprint or using Cura plugin?

When I print using USB flash drive, slicing with Cura on my PC, the end g-code takes care of the proper homing for both the print bed and the tool head.

Since I am interested into setting up octo print so I can record my print timelapses, I would love to support whenever method you got to have Octoprint working.

yvaha, 2021-12-31:
> Hi @yvaha thanks for the feedback!
>
> Just so I understand the issue with the homming after print completes: you sliced the print with the printer settings into Octoprint or using Cura plugin?
>
> When I [...]

Hi

I'm using Ultimaker Cura on raspberry pi with the plugin. Meantime, I did a little bit of research.
The generated code contains a "G28 X Y" code which only homes the print head. Without the "X Y" the printer can also home the platform. It would be nice to have an option to choose between the two possibilities.

eskeyaar, 2024-02-14:
@ronoaldo not sure how to pull my request to here?
i have merged latest changes in my main.

ronoaldo, 2024-02-14:
> @ronoaldo not sure how to pull my request to here? i have merged latest changes in my main.

Hi @eskeyaar - from your repository you should see at the home page a mention that "your fork is XX commits ahead, YY commit behind ...". This info should have links for you to be able to submit a PR.

Hope that helps!