ronoaldo/FlashforgeFinderIntegration #13: Cura 5

mtongnz, 2022-05-08:
Do you plan to make this available for Cura 5?

ronoaldo, 2022-05-09:
Hi! Yes, I plan to support the newer version, but I have limited time. I'll
take a look at the API changes for both my plugins.

In the meantime, I kindly suggest sticking with Cura 4.

If you have the time I can review any pull requests. Since I don't remember
to use much UI code, porting to the new version should not require too much
effort.

I'll try to add some test cases, build scripts and GitHub Actions to help
the process.

Thanks!

mtongnz, 2022-05-09:
Great :)

I've managed to get it mostly working in Cura 5.  I haven't tested in older versions.  And the grayscale image conversion doesn't seem to want to work.

So it's not really ready to merge but is there so hopefully my fork will be of assistance.

jcbelanger, 2022-08-02:
Just an FYI to others who can't wait.  I added support for Cura 5 with pull request https://github.com/ronoaldo/FlashforgeFinderIntegration/pull/15

ronoaldo, 2022-08-02:
Thanks to @jcbelanger - this is now suported! I'll upload to Marketplace in the next few days.

closed 2022-08-02 by ronoaldo

ronoaldo, 2022-08-02:
Uploaded to Marketplace - should be reviewed and tested by Cura team soon!