ronoaldo/FlashforgeFinderIntegration #17: Adventurer 4 format

DanielO, 2022-09-25:
Hi,
I didn't notice this plugin existed so I wrote my own but didn't get around to publish it yet.
One thing mine does is create a higher resolution PNG image (320x320) - not sure if you are interested in adding it.

The code is at https://github.com/DanielO/GXCodeWriter/blob/main/GXCodeWriter.py if you want to take a look.

(I'm happy to dual license it etc..)

ronoaldo, 2022-09-26:
Hi Daniel! Yeah, the package was missing from the Marketplace for a while after Cura 5 release. Thanks to #15 and after I managed to pass the publish step it should be working fine.

Your suggestion seems doable with little effort indeed. Do we have a printer profile available in Cura for the Adventurer already? If we do, I can detect the size/format based on that. Otherwise, we can do a Settings Toggle for the GXWriter plugin based on your code. My implementation is encapsuled into this module, and I guess it could be extended to support a larger preview image in PNG format: https://github.com/ronoaldo/FlashforgeFinderIntegration/blob/master/plugins/GXWriter/gx.py

Regarding license, I'll check but as your work is currently BSD 2-clause and mine was GPL-3, I guess both could be distributed without issues.

DanielO, 2022-09-27:
I don't think there is a printer profile yet - I am still tuning mine.
I wonder if the older printers would ignore the PNG - it is implemented in the format separately to the BMP.
The header has 3 32 bit offsets and it looks like the older format doesn't use the second one but the new one does - that is where the PNG lives.
I created a file with my plugin:
https://www.dons.net.au/~darius/xyzCalibration_cube.gx

Does it work OK on your printer? (eg the LCD screen preview)

davidt94, 2023-02-01:
Thank you both for the work on this. Was anything done about implementing Daniels solution?

thank you!