ronoaldo/FlashforgeFinderIntegration #18: Add support for Finder 3

ronoaldo, 2022-11-26:
There is a new support for the Finder 3. This issue is to track support for that new model.

Artemys59, 2022-11-27:
Hi, I am looking for FF Finder 3 parameters for Cura. Here is what I have for the moment. 

Printer settings (OK)
X 190.0
Y 195.0
Z 200.0

I still haven't figured out what are the printhead settings, feel free to suggest yours :)

I extracted Start G-code and End G-code from flashprint g-code files when slicing models.

Start G-code 
M118 X18.5 Y26.25 Z7.65 T0 (I think this line is wrong or not needed)
M140 S60 T0
M104 S200 T0
M104 S0 T1
M107
G90
G28
M132 X Y Z A B
G1 Z50.000 F420
G161 X Y F3300
M7 T0
M6 T0
M651 S255

End G-code 
M104 S0 T0
M140 S0 T0
G162 Z F1800
G28 X Y
M132 X Y A B
M652
G91
M18

Hoping someone could provide some help here to find the missing parameters. Have a nice day !

[image]

eeeebbbbrrrr, 2023-01-15:
Just a note to +1 this and I'd be happy to help test anything.

I'll try what @Artemys59 has suggested above and report back soon.

Swamp-Ig, 2023-07-31:
I think these lines in the start code need to be changed:

```
M140 S60 T0
M104 S200 T0
```

to something like:

```
M140 S{material_bed_temperature} T0
M104 S{material_print_temperature} T0
```

Otherwise the machine doesn't heat to the right temp, always goes to bed temp 60 and extruder temp 200