ronoaldo/FlashforgeFinderIntegration #21: Fixed Temperature assigned by gx script??

Phantom2200, 2023-03-21:
No matter what temperature i choose in my Cura profile, it sets to 180°C which is too low (stepper motor starts clicking because the filament is under too much pressure)

ronoaldo, 2023-03-21:
The file itself has the temperature, but it looks like the temperature is set as a floating point number, like 220.0, and before it was set as 220 without the .0 at the end. This was a change in Cura itself as far as I could tell, but I could be wrong.

Santoshsomi, 2024-02-14:
@ronoaldo i dont think there is no issue in temp. I checked with Cura 5.3.0, seems okay for me. Can you give me gcode file where you see issue? I know its a old issue but i can fix it if the issue persists because i created the definition files for finder. 

- Santosh (eskeyaar)

ronoaldo, 2024-02-14:
Hi! Yes, I can share the file with you. Allow me some time to reinstall Cura and create the file for you.

Best regards.

ronoaldo, 2024-02-14:
OK, I guess this was a bug that I never noticed but here are the results of my testing:

1. If the G code command sends the temperature as a float number (200.0 instead of 200) the Finder firmware seems to ignore that value and defaults to the lowest possible temperature
2. If the xgcode command has a temperature in it, it is used by the finder as a "warming" or initial temperature and as such it should work

The code here should parse the starting temperature (material print temperature) and use that to the header, but this is currently not done:

https://github.com/ronoaldo/FlashforgeFinderIntegration/blob/master/plugins/GXWriter/GXWriter.py#L90

Perhaps refactoring as proposed in #16 could help fix this as well.

ronoaldo, 2024-02-14:
Github is not allowing me to upload the file here, so I'm sharing from Drive:

https://drive.google.com/file/d/1agIAbxvDT47iYwhzjLXFO2VcN8m1OkJj/view?usp=drivesdk

Had to reinstall Cura to latest version.

One thing I noticed is that if the temperature GCode has a comment, Cura adds a .0 at the end of the temperature settings. I'm not sure if that even makes sense.

eskeyaar, 2024-02-15:
I see you have fixed it, but giving cura temp is not reflecting in gcode or machine? Its bizarre that float value can affect changes too..

ronoaldo, 2024-03-10:
> I see you have fixed it, but giving cura temp is not reflecting in gcode or machine? Its bizarre that float value can affect changes too..

It has been a while since I tested this and I could be wrong but if I'm not mistaken this was what I noticed. Interestingly enough, if the custom gcode has comments Cura generates temperatures with that dot, without comments it generates without the dot in the temp number. Quite odd! Perhaps some g-code flavour shenanigans.

ronoaldo, 2024-04-09:
Hi! I believe that this was fixed with the 2.1 release, also published on the Cura Marketplace. Can you guys confirm? I was able to reproduce and see the fix on my Finder.