ronoaldo/FlashforgeFinderIntegration #23: Can't select .gx in Cura 5.3.0

kingnoob1377, 2023-06-11:
Export window of Cura 5.3.0 is like this:
[image]
Whereas Cura 5.2.1 has export window like this:
[image]
I have GXWriter 2.0.0 installed.
I am not expert but doesn't plugin supporting SDK 8.1.0 supports Cura 5.3.0?

eazrael, 2023-06-24:
At least in Cura 5.3.1  it is working for me. Do you use the save to disk button after slicing?

kingnoob1377, 2023-06-28:
I do. Also note that resetting settings did not fix it.

ronoaldo, 2023-06-28:
@eazrael thanks for the help!

@kingnoob1377 thanks for the bug report. I'll try to have a look at it this weekend.

You are using Linux or Windows? Perhaps I made something in the script that is not working for the windows build. If on Linux, are you using a Flatpak or .AppImage?

kingnoob1377, 2023-08-27:
Thank you for the support. I am using Windows 11.

ronoaldo, 2023-09-06:
I was on a trip, back now. I have a Windows 11 machine, and I'll try to reproduce the issue.

kingnoob1377, 2023-11-13:
Updating to Cura 5.5.0 somehow fixed this issue. Without any other person with this bug, I don't think this is needed.

Santoshsomi, 2024-02-14:
I had same issue , uninstalled gxwriter and installed again, this time its showing correctly with 5.3.0

ronoaldo, 2024-03-10:
I'll keep this open for a while because it could be a transient plugin issue. I'll try to reproduce on a Windows machine.