ronoaldo/FlashforgeFinderIntegration #28: Release 2.1.0 on the Marketplace

d1nuc0m, 2025-01-19:
As per title, as of 2025/01/19 last package version available on UltiMaker Cura Marketplace is 2.0.0; while last release on GitHub is 2.1.0

ronoaldo, 2025-01-21:
Hi! Thanks for the notification. I was positive it was properly uploaded, but I'll try to update it on Marketplace.

d1nuc0m, 2025-01-26:
@ronoaldo thanks, opening [Flashforge Finder printer Integration](https://marketplace.ultimaker.com/app/cura/plugins/Ronoaldo/FlashforgeFinderIntegration) still reports:

```
Package version
2.0.0 - last update on 21-09-2022
(...)
Release notes
- Drop support for Cura 4.5 and earlier. by @ronoaldo
- Support for Cura 5.1.0 (SDK 8.1.0) by @jcbelanger
```
And same in Cura internal search:

[image]